frangoteam/FUXA #846: New control to FUXA

AlexandruAndrei3, 2023-06-14:
Hello, i am trying to implement https://github.com/nicolas-van/steelseries into fuxa. Like it was specified in a previous post i used iframe as a model. 

In svgIframeContainer.innerHTML i used this template declared in a variabile.

  static radial =  `
        <script type="module" src="https://cdn.jsdelivr.net/npm/steelseries@2.0.7/dist/steelseries.bundled.min.js"></script>
        <steelseries-radial size="200" value="73" maxValue="100" gaugeType="TYPE3" frameDesign="GLOSSY_METAL" pointerColor="ORANGE" knobStyle="SILVER" thresholdVisible maxMeasuredValueVisible foregroundType="TYPE4" tickLabelOrientation="HORIZONTAL" useOdometer></steelseries-radial>`;

The control is displaying fine but i cannot access the proprieties like value outside the template.
Also i cannot find where size for iframe control are defined, is seems its 240px x 170px.

How i can access the values of the template?

AlexandruAndrei3, 2023-06-15:
Or if it is possible to use default angular html templatel for innerTEXT.

unocelli, 2023-06-18:
Hi, currently the default size of own controls is defined as 240x180 and cannot be changed.

unocelli, 2023-10-11:
Sorry, I correct myself, you can define the size, as an example look at the last added control GraphPieComponent

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli

jay-jjwu, 2024-08-05:
> Hello, i am trying to implement https://github.com/nicolas-van/steelseries into fuxa. Like it was specified in a previous post i used iframe as a model.
>
> In svgIframeContainer.innerHTML i used this [...]

@AlexandruAndrei3 , hi, did you successfully add steelseries into fuxa controls?