frangoteam/FUXA #854: Adding Support for MQTT with TLS Using Certificates in FUXA

eduardojvieira, 2023-06-19:
Hello frangoteam,

I've been using FUXA in my recent project to visualize and render data using MQTT. I really appreciate the functionality it offers; however, I have encountered a situation where I need to establish a secure MQTT connection using SSL certificates (CA cert, Client cert, and client key).

Currently, FUXA supports MQTT connections with authentication via username and password. While this works fine in many situations, it falls short when advanced security is required and secure connections need to be established via SSL/TLS using certificates.

The underlying MQTT.js library which FUXA uses for MQTT connections already has support for TLS connections (see example here: https://github.com/mqttjs/MQTT.js/tree/main/examples/tls%20client). Therefore, I believe it would be a valuable feature addition to allow users to specify CA cert, Client cert, and client key within FUXA to take advantage of the secure MQTT with TLS provided by MQTT.js.

This enhancement would not only increase the level of security in the application, but also broaden the scope of projects that could utilize FUXA.

Please consider this feature request for a future release. I believe it would be a great asset for many developers dealing with secure IoT applications and more.

Thank you for your time and consideration.

Best regards,
Eduardo

unocelli, 2023-06-21:
Hi, doesn't look complicated, how many and what types of certificates should the GUI allow you to configure?

eduardojvieira, 2023-06-21:
Normally three files are needed to establish the connection using TLS:

The CA certificate (CA Root)
The client certificate
client private key

unocelli, 2023-07-09:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-07-09 by unocelli