frangoteam/FUXA #856: Editor theme switch

budou0608, 2023-06-21:
Version: FUXA-1.1.14-1230

client/src/app/header/header.component.html
``` html
<div class="header-right" *ngIf="editorMode">
    <div class="header-help">
        <button mat-button title="{{'header.help' | translate}}" [matMenuTriggerFor]="helpmenu" class="main-btn">
            <mat-icon aria-label="HELP">help_outline</mat-icon>
        </button>
        <mat-menu #helpmenu="matMenu" class="leftbar-item-menu header-menu" yPosition="below" [overlapTrigger]="false">
            <button mat-menu-item (click)="onShowHelp('help')">{{'header.help-tutorial' | translate}}</button>
            <mat-divider class="menu-separator"></mat-divider>
            <button mat-menu-item (click)="onShowHelp('info')">{{'header.help-info' | translate}}</button>
        </mat-menu>
    </div>
    <div class="header-theme">
        <button mat-button title="{{'header.theme' | translate}}" class="main-btn" (click)="onChangeTheme()">
            <mat-icon>invert_colors</mat-icon>
        </button>
    </div>
</div>
```
client/src/app/_services/theme.service.ts
``` ts
import { DOCUMENT } from '@angular/common';
import { Inject, Injectable } from '@angular/core';
import { THEMES } from '../_config/theme.config';

@Injectable({
    providedIn: 'root'
})
export class ThemeService {
    constructor(@Inject(DOCUMENT) private document: Document) { }

    static ThemeType = { Dark: 'dark', Default: 'default' };

    setTheme(name = ThemeService.ThemeType.Dark) {
        if (!THEMES[name]) {
            name = ThemeService.ThemeType.Dark;
        }
        // name = ThemeService.ThemeType.Dark;
        const theme = THEMES[name];
        Object.keys(theme).forEach((key) => {
            this.document.documentElement.style.setProperty(`--${key}`, theme[key]);
        });

        const body = document.getElementsByTagName('body')[0];
        body.classList.remove('dark-theme');
        if (name === ThemeService.ThemeType.Dark) {
            body.classList.add('dark-theme');
        }
    }
}
```

unocelli, 2023-06-21:
Hi, if you activate this option then you have to manage and test it, which increases the workload.

budou0608, 2023-06-23:
OK！I've already tested it

budou0608, 2023-06-23:
What is the bug fix submission process

unocelli, 2023-10-11:
Hi, sorry for the delay...I will add a switch as beta function in the next version.

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli