frangoteam/FUXA #857: CompactLogix PLC - Node PCCC plugin

jpomo, 2023-06-22:
I'm attempting to use a compactlogix plc (1769-L24ER-QB1B) with the Node PCCC plugin.

I'm trying to understand how the routing should be configured. The PLC show itself in slot "0" but there is no slot # for the ethernet port, as it is embedded. No combination of routing values has worked so far (0,0), (0,1), etc.  and results in the following output for the connection (attached below)

I assume this is because node PCCC was never tested on this particular model.

Any advice is welcome.

[image]

SFTP-ChZeman, 2024-02-28:
I've been using Node-RED to collect data from PLCs and there are two plugins... PCCC and Ethernet/IP. The Ethernet/IP plugin for FUXA appears to actually be PCCC and not Ethernet/IP, at least in terms of how you need to talk with a Micrologix vs a ControlLogix. PCCC expects old style addressing (ie., I:0/0, B3:0, etc.) and Ethernet/IP is reading tags.

jpomo, 2024-02-29:
> I've been using Node-RED to collect data from PLCs and there are two plugins... PCCC and Ethernet/IP. The Ethernet/IP plugin for FUXA appears to actually be PCCC and not Ethernet/IP, at least in [...]

Thanks for the reply! I found that FUXA works ok running a Modbus TCP client on the CompactLogix. I hope we will one day FUXA will get a true Ethernet/IP Plugin.

SFTP-ChZeman, 2024-02-29:
I sure hope so, too! I'm trying to avoid MODBUS because I want to read strings. I'm actually using a Productivity 1000 (Automation Direct) controller so I'm working on MQTT since Ethernet/IP won't work in this situation. I'm just having a little trouble with MQTT. If it's not one thing, it's another. LOL

unocelli, 2025-05-25:
Hi, let's move this discussion to #1213 as suggested by @jpomo

closed 2025-07-24 by unocelli