frangoteam/FUXA #858: Can't run built app

sokol07, 2023-06-28:
Hi,
First, I should mention that my knowledge about JS and all things connected to it is extremely basic so it is very likely that I'm missing something obvious for someone who is used to it.
I have dodwnloaded Fuxa, created an app and I want to build it.
I had some trouble running `ng build --prod` (it required newer NODE), however I managed to update everything and got the build command passing.
It got me following output:
```
C:\Users\Tomasz\Desktop\FUXA-1.1.12\FUXA-1.1.12\client> ng build --prod
Option "--prod" is deprecated: Use "--configuration production" instead.
\ Generating browser application bundles (phase: setup)...Processing legacy "View Engine" libraries:
- angular2-draggable [es2015/esm2015] (https://github.com/xieziyu/angular2-draggable.git)
Encourage the library authors to publish an Ivy distribution.
√ Browser application bundle generation complete.
√ Copying assets complete.
√ Index html generation complete.

Initial Chunk Files           | Names         |  Raw Size | Estimated Transfer Size
main.e30a4fa9967c3625.js      | main          |   5.27 MB |
1.26 MB
styles.ec6b5cd5654140fe.css   | styles        | 203.92 kB |                19.02 kB
scripts.ebd98081df738ac3.js   | scripts       |  42.33 kB |                17.05 kB
polyfills.15e52e6eacaabf13.js | polyfills     |  33.16 kB |                10.64 kB
runtime.baefb0096c638cd1.js   | runtime       |   1.71 kB |               877 bytes

                              | Initial Total |   5.54 MB |
1.30 MB

Build at: 2023-06-28T10:01:56.365Z - Hash: ae70fe9b00d7da00 - Time: 120043ms

Warning: C:/Users/Tomasz/Desktop/FUXA-1.1.12/FUXA-1.1.12/client/src/environments/environment.client.ts is part of the TypeScript compilation but it's unused.
Add only entry points to the 'files' or 'include' properties in your tsconfig.

Warning: C:/Users/Tomasz/Desktop/FUXA-1.1.12/FUXA-1.1.12/client/src/environments/environment.demo.ts is part of the TypeScript compilation but it's unused.
Add only entry points to the 'files' or 'include' properties in your tsconfig.
```
I guess this means that the app was build correctly? However, If I try to run my app (I guess running it means opening the index.html file from the dist folder?) it is stuck on the loading screen with following errors in browser console:
[image]

I tried the most obvious thing (copying the files to `C:/`) - the "file not found" errors disappeard (not all, main(...).js and runtime(...).js remained even though these files were copied to the C:/ folder too) but the CORS errors remained and the app was stuck on the loading screen anyway.
What am I missing? How should the app be build and run? Is there a manual/docs about that anywhere?

Best regards and thanks in advance!

unocelli, 2023-06-28:
Hi,
- if you do `npm start` in client folder you are starting the frontend (angular) web server for debugging, that are listening at port 4200.

- if you do `ng build --configuration=production` in client folder you are build the production frontend, that will be used from FUXA server at port 1881.

in any case you need to started the FUXA server for the communication (debug or production)

sokol07, 2023-06-28:
Hi,
thanks for answer!
It is very likely that I'm missing some concept here, I'm not very familiar with angular and, generally speaking, JS. Sorry for the avalanche of questions but I'm trying to organise mess in my head a bit...

1. Do I understand correctly - this lets me, for example run the server and client on separate machines?
2. What is the difference between production and debugging frontend? (not from theoretical but from practical point? debugging works locally and production works remotely? what is the use case and procedure for each of them?)
3. Is there any manual/docs about how to run the client and server?
4. How should all this be started?
      a) I do `npm start` in `/server` to start the server on server machine and leave it running?
      b) And what about the client? I place the client files (made by `ng build`) on the client machine and that lets me display data from the server? Do I understand correctly that, for example, these client files (from `/dist` folder) can be run using an nginx server?
      c) If the above conception is ok, how is the client-server connection configured?
5. What should I do if I want to run Fuxa locally? (Fuxa is running on one PC, the data source is connected over modbus, the operator is accessing only this particular PC) Do I just run `npm start` in `/server` and use the interface it provides on localhost (I don't use the `/client` folder in any way)? 

Generally speaking: what is the point of separating the client software and running it separately? Is this the matter of multi-user access? If I run `npm start` in `/server` I can access Fuxa from other devices in the network (by http://<ip>:1880), even if they don't run client software, just a browser, so in what scenario is the client needed?

Sorry if my questions are missing the point, I hope you will be able (you will have patience) to fix all my misconceptions on this topic...

unocelli, 2023-06-30:
Hi,
1. Yes (client is your browser)
2. In a production frontend, the code is typically optimized for performance and file size. This involves minifying and compressing the JavaScript, CSS, and HTML files, removing unnecessary code and comments, and optimizing the code for faster execution.
3. [wiki](https://github.com/frangoteam/FUXA/wiki) and https://en.wikipedia.org/wiki/Web_application
4. You should only start the server and connect it with a browser, you don't need to start any client
5. Start the server and connect it with the browser to http://localhost:1881

https://en.wikipedia.org/wiki/Web_application

sokol07, 2023-06-30:
Thank you very much, this clarified things for me. I'm still a bit (but far, far less!) confused.
> 4. You should only start the server and connect it with a browser, you don't need to start any client

Is the building the production frontend (https://github.com/frangoteam/FUXA#to-build) used for the mentioned minifying ond optimization?
Which means - after creating my project I run `ng build --prod` in `/client` and then run everything using `npm start` in `/server`?
If not, what is the use case of the mentioned `ng build` in `/client`? (I guess this was my question and misunderstanding from the beginning...)

closed 2023-08-16 by unocelli