frangoteam/FUXA #859: Hello! I would like to inquire about two questions about charts. Thank you for your time.

gospeell, 2023-06-29:
1.How to transfer different parameters and call the same chart like the method of "How To use same view"? This makes it easy to view the trend of each parameter.

2. Pop up chart, clicking the button above the chart will close it. Actually, I want to adjust the time of the chart, I don't know if this is a bug.

I hope you can understand the problem I described. It must be the result of Machine translation. Thank you for your time.
[image]

unocelli, 2023-06-30:
Hi, 
1. Currently is not supported.
2. is a bag...thanks for the report.

killerbeat12, 2024-05-20:
Yes, i am also looking for this feature would be nice! keep up the good work :)!

unocelli, 2024-07-14:
do you have any suggestions as to how this should work?

killerbeat12, 2024-07-16:
i think easiest way is scripting, assigning tags to placeholders  in a script (internal tags)?

closed 2025-07-24 by unocelli