frangoteam/FUXA #864: How to use WebAPI?

Doboo, 2023-07-14:
Hi;
I noticed that the latest released version mentions new features: 'New WebAPI' API/getTagValue 'and' API/setTagValue '.

But how can someone help with using this WebAPI?

Doboo, 2023-07-20:
Thanks a lot!

Doboo, 2023-07-20:
I have tested and there is no problem with “getTagValue”  . But when use "setTagValue", the numerical value has already been written, but the prompt message is: error: "not_found" Message: "tag id not found: t_5588a151-0dbf4300; t_080f5955-0fd54761;
I'm a bit confused.

Doboo, 2023-07-23:
And another question: 
How can I use these  api  wiht token?

unocelli, 2023-07-25:
Hi, currently is not supported, we have a apiKey planned.

unocelli, 2023-08-08:
@Doboo about the error that you have reported "not_found" Message: "tag id not found: t_5588a151-0dbf4300; t_080f5955-0fd54761;
I suppose that you are try to write in tags that are not found, please check your code.

Doboo, 2023-08-10:
I tested the simultaneous writing of two points, and the values have been successfully written. But the prompt 'tag id not found' is just a bit strange, but it does not affect the use of the function.

unocelli, 2023-08-17:
Hi, thanks for the report, We can reproduce it and we will fix it

closed 2023-10-11 by unocelli

RajatDas2020, 2024-08-01:
Hi I want to get and post tag values, from Rest API testing tool. How to expose this tag id "id": "t_2f47264f-b9e4451d" to the outside fuxa ?

unocelli, 2024-08-01:
@RajatDas2020 look at #650

RajatDas2020, 2024-08-02:
Thank you, but what is the complete url for post , do we need to specify tag in the url itself like get ?
POST /api/setTagValue

unocelli, 2024-08-17:
No you don't need in the url but as parameter