frangoteam/FUXA #865: Shape settings for each bit from bitmask

IonuDragomir1, 2023-07-14:
Hi,
Let's say that I have the following scenario:
I have a pump controlled by a frequency converter. Depending on it's status word (SW), I have to change pump's shape settings. In the below image is represented each SW's bit specification.

[image]

So, for example, the pump should be green if bit 11 of SW is true (1). **The same pump** should be colored red if one of  4, 13 or 14 bit of  SW is true (1).

As you can see in the below image, I cannot set a color for each selected bit.

[image]

It'd be great to have the possibility of assigning a value(true or false) for each selected bit from bitmask).
Assigning a value to entire SW is not a solution because it's constantly changing.

concept10, 2023-07-21:
Hello,

I'm generally curious to what drive manufacturer you're using in this example?

Thanks

IonuDragomir1, 2023-08-09:
Sorry for the delay. It's a Danfoss FC202.