frangoteam/FUXA #866: svg import issue

patson12, 2023-07-14:
Hii,

i am uploding some svgs, but it didnt reflet properly .. attacyhing some svg .. can you plz check it?
[image]

unocelli, 2023-10-13:
In the last version there is a new add image link that solve the problem.
[image]

guiview, 2023-10-27:
[image]

This image import working show but not perfect import this image

unocelli, 2023-10-31:
I can't reproduce your issue
[image]

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli