frangoteam/FUXA #868: Switch button changes its size while being used.

Bundyra, 2023-07-19:
Hello, today i found the issue with switch button. If you change the button's default size of 50x28 to any other size, then pressing the button causes it to change its size/position (not sure which one). Example: i changed buttons size to 30x18 and this what happens after pressing it:
[image]

unocelli, 2023-07-23:
Hi, We can't reproduce it...can you send me an example project?

Bundyra, 2023-07-25:
Sure, I sent an email with an attachment.

unocelli, 2023-07-25:
Hi, Thanks, I can reproduce it and we will fix it as soon as possible.

WeiFengHong, 2024-07-19:
Hi, is there any progress on this issue? There is still a size change problem when switching status.