frangoteam/FUXA #870: What are the minimum system requirements?

jingshui127, 2023-07-19:
I have an Ubuntu system, 4 core A9 processor, the main frequency up to 1.4GHZ; 32-bit high-speed DDR3 memory 512M. Can I install and run FUXA?

Bundyra, 2023-07-19:
I run it on Orange Pi Zero2 with no problem.

jingshui127, 2023-07-20:
root@he:~# docker pull frangoteam/fuxa:latest
Error response from daemon: manifest for frangoteam/fuxa:latest not found
root@he:~# docker pull frangoteam/fuxa:latest
Error response from daemon: manifest for frangoteam/fuxa:latest not found

jingshui127, 2023-07-20:
[image]

from :  jessejamescox/fuxa:latest

jingshui127, 2023-07-21:
[image]

concept10, 2023-07-21:
FUXA should run only any device that nodejs version supports. You can't guarantee any acceptable level of performance beyond testing with those nachine specs in reality.

jingshui127, 2023-07-22:
Use docker.
Cannot install from official. It can only be taken from a third party.

jingshui127, 2023-07-22:
> Use docker.
> Cannot install from official. It can only be taken from a third party.

So I can't use the latest version.

unocelli, 2024-05-22:
I’m closing this issue since there is nothing feasible and due to inactivity.

closed 2024-05-22 by unocelli