frangoteam/FUXA #871: Modbus readMemory error

pwcpp, 2023-07-19:
I have two Modbus RTU devices connected via Modbus Gateway (Modbus RTU to Modbus TCP).  Each device works fine on its own (disable another device).  But if both devices are enabled, Modbus "readMemory error" will occur frequently which result in not very stable communication .  Increasing poll or activating fragmented still doesn't work.

Before testing FUXA, both RTU devices worked well with other Scada software.   Any suggestions (e.g. timeout settings, which I'm not sure where to find)?  Thanks.

rvbatista, 2023-07-19:
Hello,

Both devices are below the same gateway right?

Today fuxa doesn't have any kind of serialization to devices sharing the same gateway, this is a feature that I was working, but was not able to finish yet, I believe that we already have a issue related with this behavior.

pwcpp, 2023-07-20:
THX for reply.  Yes, both devices are below the same gateway.  

Sorry for not searching for the same issue before.  Pls. close it.  THX

closed 2023-08-16 by unocelli

liulujian00, 2023-09-21:
I have the same problem and this problem is still not solved；
I use two laptops to create a device as a modbus host on fuxa, and one of them is a slave simulated with modbus slave software.

The log is as follows
cat server/_logs/fuxa1.log 
2023-09-21T03:27:31.234Z [error] 'modbus-rtu' _readMemory error! Error: Data length error, expected 9 got 3
2023-09-21T03:27:32.224Z [error] 'modbus-rtu' _readMemory error! Error: CRC error