frangoteam/FUXA #872: How to improve the issue of management page lag when the system tag exceeds 5000?

gospeell, 2023-07-20:
Hi, I have an idea to use FUXA on projects with more than 5000 system tag . Currently, during simulation testing, there was a serious lag in the management page at 5000 system tag . How can I improve this problem?

unocelli, 2023-07-25:
Hi, we have activated a option that have to be test. you find it into the master branch.
Let me know if it works.
[image]

hxjackcn, 2023-07-25:
Node.js cannot be multi-threaded. Let me test it, 5000 points, for opcua, set it to refresh in 30 seconds. If the refresh time is too small, it won't be able to handle it.

hxjackcn, 2023-07-25:
This is determined by the mechanism of Node.js itself. My current solution is that I build multiple sites, split the points, each site does not exceed 2000 points. My projects generally have around 30 million points, or even more.

hxjackcn, 2023-07-26:
The running stability is not bad, I have a project, I have used fuxa for more than 2 years, and it is still running stably.

gospeell, 2023-07-26:
> Hi, we have activated a option that have to be test. you find it into the master branch. Let me know if it works. [image]

 Okay, arrange testing as soon as possible

gospeell, 2023-07-26:
> This is determined by the mechanism of Node.js itself. My current solution is that I build multiple sites, split the points, each site does not exceed 2000 points. My projects generally have around 30 million points, or even more.

Hello, are multiple fuxa deployed on the same computer?

hxjackcn, 2023-07-31:
It's all right, I usually deploy multiple sites on one computer and deploy multiple computers.

gospeell, 2023-08-02:
Hi, I have tested using the new version and now the report is as follows:

1. 4991 system tag tested

2. There is still a problem with stuck backend settings, mainly manifested in the device page and save menu options being stuck and unresponsive

gospeell, 2023-08-02:
> It's all right, I usually deploy multiple sites on one computer and deploy multiple computers.

Do you have any tutorials in this area？

unocelli, 2023-08-12:
Hi, sorry for the delay, do you mean that without going into edit mode and having disabled broadcasting of all tags, the views work correctly? and the problem is still if you go to edit the device?

unocelli, 2024-05-22:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-05-22 by unocelli