frangoteam/FUXA #878: ModbusTCP / ModbusRTU Configuration

tomas-kucera, 2023-07-28:
Hi there, I am struggling with the configuration of the ModbusTCP / ModbusRTU Tags.

The dialog allows me to set InputRegisters / Holding Registers within the range of 300001 - 365536 (400001 - 465536 resp.). But the specs for my device say:

Input register(s) address       Purpose
0-15                                         data unit 1
16-31                                       data unit 2
...

Do not take me wrong, I understand the ranges (as they come from Modbus specs) but mostly the device documentations do not use this notation and it just adds extra layer of complexity to the configuration.

I had tried to test it with the modbus-serial library and there is absolutely no problem to specify and execute that I want to readInputRegisters from address 0, etc. So it is not a limitation / requirement of the library but the FUXA dialog.

Why the dialog does not specify only that I want to perform:
FC01 - Read Coils
FC02 - Read Discrete Inputs
FC03 - Read Multiple Holding Registers
FC04 - Read Input Registers
...

And the allow to enter any addres in the range of 0 - 65535 and in case of multiple registers / coils / inputs the number of them.

I think that for most people this would be a lot more understandable and convenient.

Your thoughts on this? Thanks!

unocelli, 2023-08-06:
Hi, i agree with you that modbus addressing is a bit confusing and there are different interpretations of how to display the address and it also depends on the user. probably our interpretation of how to display the address settings is not the happiest. let's see if we can find a consensus for improvement...thanks for the input.

tomas-kucera, 2023-08-13:
Hi, great! I would be happy to provide some feedback and wireframes based on my experience from working with Modbuses.

rvbatista, 2023-11-19:
Unfortunatelly Modbus specification don't specify this kind of information.
What I already saw in some softwares is to have IEC61131-3 (%M, %MW addressing), index (0-65535) and 'legacy' (0000x, 1000x, 4000x, etc...) options to act as a mask for value of the address field.
But what I think that is better and simple, it's only index, always starting from 0, that is exactly what will be on the wire.

unocelli, 2024-10-27:
I’m closing this issue since there is nothing feasible and due to inactivity.

closed 2024-10-27 by unocelli