frangoteam/FUXA #881: How to fuxa read data int64 from Modbus TCP

UNiItCode, 2023-08-03:
Hello, i try read data from modbus sim, data int64. Fuxa tag property type have not  int64. How can I fix this problem? 
Thank you.

unocelli, 2023-08-06:
Hi, you have to add it in [backend type definition](https://github.com/frangoteam/FUXA/blob/master/server/runtime/devices/modbus/datatypes.js) and add in [frontend to ModbusTagType](https://github.com/frangoteam/FUXA/blob/master/client/src/app/_models/device.ts)

UNiItCode, 2023-08-12:
Thanks for the reply. I have Fuxa Pro but don't have file device.ts in folder client.

Can you give advice?

unocelli, 2023-08-12:
Hi, for support on Fuxa Pro you can contact [frangoteam](https://frangoteam.org/)

unocelli, 2023-08-16:
Hi, before to add in Pro version we have to make some test. There is a problem where in javascript the Int64 is a BigInt type and it is not managed in fuxa, it has to be converted into a number type and can lose some precision...I make the change in dev branch in github, please test it and give me a feedback.

unocelli, 2023-10-11:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2023-10-11 by unocelli