frangoteam/FUXA #883: Bug while running multiple scripts

IonuDragomir1, 2023-08-04:
Hi,

I was facing the following bug when using scripts. On my FUXA project, I have 3 scripts (S1, S2, S3). Each of them is running at 5 seconds.. After editing and saving S1, I have noticed the none of the scripts were working.  I have spent some time to figure it out the cause. I made a mistake which was causing an error. When I edited S1, I had a ";" instead a ",".  The problem is that this error stopped the good working process of S2 and S3 which had no error. Normally, only S1 should not be working if an error occurs. If S2 or S3 have no error, their execution process should not be interrupted by S1's error.

unocelli, 2023-08-06:
Hi, Indeed, ";" instead a "," is a bad error that can have unexpected consequences in nodejs, tools such as linters should be added.

robsori, 2024-03-21:
It would be very helpful to implement this "linter" function because when a function contains several code instructions, the probability of making a syntax error is very high and I noticed that the whole project gets blocked. I think it would be a priority facility for Fuxa.

closed 2026-11-10 by unocelli