frangoteam/FUXA #888: How to read coil status on Modbus TCP?

IonuDragomir1, 2023-08-08:
Hi,
I want to read, using Modbus TCP, coil's status from Schneider PLC. Coil's address is Q0.4. In Fuxa, I have created a tag, using Coil Status register, type bool. What address should I use to read coil's status? I have tried address 3, 4 and 5, but none of them returns coil's status.

rvbatista, 2023-08-09:
I believe that these addresses are not mapped to modbus in Schneider PLCs, which exact model are you using?
Try to do the same with modscan or another modbus test software.

IonuDragomir1, 2023-08-09:
@rvbatista, I am using TM221CE24R model.
I have used, as you suggested, modscan. Scanning 100 coils (addressing from 1 to 100), I've noticed that coil 41 is changing its value as Q0.4 does. I put the same address (41) in FUXA and it works as expected. Now the question is what's the relation between Q0.4 and address 41?

rvbatista, 2023-08-09:
Great, I didn't know that the outputs are mapped to Modbus in M221.

IonuDragomir1, 2023-08-09:
What's strange is that I changed Q0.4 with Q0.5 and still coil 41 is changing it's value accordingly with Q0.5 value. Shouldn't have been another address instead of 41?

rvbatista, 2023-11-19:
> What's strange is that I changed Q0.4 with Q0.5 and still coil 41 is changing it's value accordingly with Q0.5 value. Shouldn't have been another address instead of 41?

You figured out if what you are reading are really related to the digital outputs? Searched a little the documentation and didn't found this relation.

IonuDragomir1, 2023-11-20:
Hi,
@rvbatista, unfortunately I didn't figure out. The TM221CE24R PLC has 8 DO (DigitalOutputs). On it can be added DO extensions. A DO extension has outputs with address from Q1.0 to Q1.31. I really don't know how I can read on FUXA Q0.4's or Q1.13's status. It's not clear what address should I use to read coil's status.
Maybe @unocelli or somebody else can help with this issue.

rvbatista, 2023-11-21:
@IonuDragomir1 the issue is related to the hardware found M221 function code list.

[image]

Believe that this issue can be closed.

IonuDragomir1, 2023-11-21:
@rvbatista thanks for the reply!
So the only way to read coil's status is to transfer its value to a %M internal bit and read it on FUXA. 
Maybe DOs from Schneider's premium PLC class can be read directly with FUXA.  Sometime I'll give it a try.

rvbatista, 2023-11-21:
> @rvbatista thanks for the reply!
> So the only way to read coil's status is to transfer its value to a %M internal bit and read it on FUXA. 
> Maybe DOs from Schneider's premium PLC class can be read directly with FUXA.  Sometime I'll give it a try.

Yes, to read outputs we need to move the %Q to %M or even %MW.
It's good to find similar table for PREMIUM.

unocelli, 2024-05-22:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-05-22 by unocelli