frangoteam/FUXA #895: Embedded control request

Beyond2324, 2023-08-18:
Hello, I want to know how to embed the new charts control in fuxa. I observed that the client encapsulates Fuxa-editor.min.js. Can I give the source file

unocelli, 2023-08-18:
Hi, to add your own control you don't need fuxa-editor.min.js. You have to follow how one of the latest controls has been implemented: iframe (html-iframe) and Table (html-table). Don't forget to share your control (PR)

closed 2023-08-18 by unocelli

Beyond2324, 2023-08-18:
> 您好，要添加您自己的控件，您不需要 fuxa-editor.min.js。您必须遵循最新控件之一的实现方式：iframe (html-iframe) 和 Table (html-table)。不要忘记分享您的控制权 (PR)

Hello, I have seen the use of iframe in the project, but I can only add new contents by adding connections, and I cannot connect new controls with server-side data

unocelli, 2023-08-18:
Hi, I think you are making a mistake or forgetting something... but I cannot guess it 😉

jay-jjwu, 2024-08-05:
> Hi, I think you are making a mistake or forgetting something... but I cannot guess it 😉

@unocelli, i also want to develop and add new controls, besides to refer to the latest controls like  iframe (html-iframe) and Table (html-table), is there any tutorial or guide for this? i think this could be a very common demand in community.
Thanks!

unocelli, 2024-08-17:
@jay-jjwu we have planed