frangoteam/FUXA #904: Where to view  ext-bundle.min.js  source code， non confusing source code

gitzhangliang, 2023-08-28:


gitzhangliang, 2023-08-28:
I want to customize a component

unocelli, 2023-08-28:
Hi, to add or customize a control you have to follow how one of the latest controls has been implemented: iframe (html-iframe) and Table (html-table). Don't forget to share your control (PR).

closed 2023-08-28 by unocelli

gitzhangliang, 2023-08-29:
I just want to know how to implement the overall process of a control。I find some source code of html-table control  in  ext-bundle.min.js,  just like

`window.extOwnCtrl = function() {
    window.svgEditor && svgEditor.addExtension("own_ctrl", function(c)
`
so I want to know Where to view ext-bundle.min.js source code

unocelli, 2023-08-29:
In fact with the latest addition to the library it is possible to define your own control without having to modify ext-bundle.min.js. 
 I am currently developing a new control and use the same strategy.