frangoteam/FUXA #907: view bug

kekebanban, 2023-09-04:
multiple people edit at the same time, there are two MainView occur (don't know the reason), and can't delete, these two views are not clickable now, how can I delete them
[image]

IonuDragomir1, 2023-09-04:
Hello,
While I was working with my colleagues on the same project, have noticed that some screens were doubled as you shown in the above picture. To delete the screen you have to edit project's json file and delet from there the doubled screen. Also, in the project's appdata folder, from project.fuxap.db file, you have to delete the doubled screen.

kekebanban, 2023-09-04:
@IonuDragomir1 
I delete them by changing project.fuxap.db file directly, thank you!

unocelli, 2023-09-07:
@IonuDragomir1 Hi, have you found how you can reproduce it?

IonuDragomir1, 2023-09-08:
Hi,
Please read my last comment on [#757](https://github.com/frangoteam/FUXA/issues/757). Though it also might be another scenarios when this error occurs. 

@kekebanban, did you opened an older JSON file on a newer fuxa version? Or did project's screens contain large/multiple images? Trying to figure out the cause of your error.

Jue07, 2024-01-10:
> @IonuDragomir1 I delete them by changing project.fuxap.db file directly, thank you!

Hi..how to edit this? where to find this file? Thanks

IonuDragomir1, 2024-01-10:
@Jue07 
Hi,
The file can be found in project's structure: ...server\_appdata
The file can be opened using SQLiteStudio.

closed 2025-10-03 by unocelli