frangoteam/FUXA #910: From the perspective of large-scale use, there are two things that need to be improved to make it more efficient.

hxjackcn, 2023-09-14:
1、In projects, pages that are similar to each other are often created, or multiple groups need to be repeated on the same page. At present, it is possible to achieve page cloning and group copy within the page. But it lacks the following functionality: batch search and replace for all element data sources.

2、The pictures on the page do not need to be placed in the database, but are loaded into the folder through links. Put it in the data, loading is too slow.

hxjackcn, 2023-09-23:
The picture problem has been solved, it’s great.

unocelli, 2023-10-22:
About the 1° comment, you can change the tags binding by use 'Edit bind of Tags' in context menu.
[image]

or via 'View item list'
[image]

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli