frangoteam/FUXA #911: [BUG]web API   data Data misalignment

adicstana, 2023-09-17:
"Data path: redis > restful > fuxa web API. When reading the data, the elements in the JSON array may not be in a fixed order, resulting in misalignment of the data points assigned to fuxa."
[image]

[image]

As shown in the figure, the value of AI-Humi in the JSON array is 61.8, but it has been wrongly assigned as 190 in fuxa.

adicstana, 2023-09-20:
thanks for your response! You probably cannot  reproduce my issue now. 
1) The order is definitely correct when connecting to the API for the first time and retrieving the data.
2) After running for some time, if the order of elements within the JSON data received from the API changes, it will cause misalignment on the fuxa side.
3) My temporary solution is to sort the data after reading it from Redis, in order to fix the data order and avoid any issues.

GitTurboy, 2023-10-08:
I found this issue too. and if there are new tags added in the webapi source, how to add them to the device?