frangoteam/FUXA #912: Installation failure

Noobert01, 2023-09-17:
Hello all,

I am looking for a free HMI for my Raspberry Pi and came across FUXA. 
I followed the instructions from the wiki for the installation. Unfortunately, the installation always fails at the same point with error no. -13. I have run the installation as sudo and enabled the "User" folder with permissions for everyone. 
I can't reach the page with http://localhost:1881.

Does anyone know what this could be?

**Screenshot**
https://github.com/frangoteam/FUXA/assets/145294314/c5606753-6b5f-4f7a-bc8f-1b6531d79133

**Environment**
 - FUXA version: newest
 - Node.js version: 14.21.3
 - npm version: 6.14.18
 - Platform/OS: Raspberry Pi 4 with Raspberry Pi OS with desktop 32bit
 - Browser: Chrome

Noobert01, 2023-09-19:
Yes, but I always come to that point.

unocelli, 2023-09-19:
how did you install it? docker, npm or from code?

Noobert01, 2023-09-19:
I use npm 6.14.18

unocelli, 2023-09-19:
Can you try to install it from code?

Noobert01, 2023-09-20:
How do I do that? I just dolleres the instructions from the wiki

Noobert01, 2023-09-20:
*followed

Noobert01, 2023-09-20:
I managed to install FUXA from code and it works. I am used to work with Windows so this is new for me.

Thank you for the support.

How do I put FUXA in autostart on the Raspberry Pi. I tried to add it to /etc/rc.local but this did not work.

Can you help me?

unocelli, 2023-09-29:
Hi, 
I would ask ChatGPT

closed 2023-09-29 by unocelli