frangoteam/FUXA #920: Bug saving a FUXA project

IonuDragomir1, 2023-09-29:
Hi,
I have the following scenario:
The user's working on project A several hours. At the end, the project A is successfully saved by clicking the Save button. Then project B is opened. When project A is reopened, all saved work is lost. 
To save project A, prior opening project B, the user should've clicked Save As button. So it'd be useful that, when project B has to be opened, the user should be forced to Save As project A. Otherwise, the user thinks that project A is saved when Save button is clicked but actually it's not if another project is loaded.
Another suggestion would be to rename Save As button with Export. In my opinion is more intuitive Export than Save As.

unocelli, 2023-10-01:
Yes, good suggestions. I prefer the version of asking the user if he wants to overwrite the current project or save it (with Save As...)

lewsut, 2025-02-07:
I agree just lost a month of noodling due to a wrong click... Now I'm sad :(