frangoteam/FUXA #921: [BUG] Record historical data in chart and table

Noobert01, 2023-10-01:
Hello all,

I am currently trying to record historical data in a chart and a table. However, I do not get anything displayed. I would like to use SQLite for this.

Can anyone help me? Is there anything I need to be aware of? I have already reinstalled FUXA but without success. I am new in this area.

I use a Raspberry Pi with Raspberry Pi OS and the latest version of FUXA. 

Many greetings

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version: v1.1.14
 - Node.js version: 14.21.3
 - npm version: 6.14.18
 - Platform/OS: Raspberry Pi 4 with Raspberry Pi OS with desktop 32bit
 - Browser: Chrome

unocelli, 2023-10-01:
Hi,
you have to enable registration settings to the tags of chart
https://github.com/frangoteam/FUXA/issues/611

unocelli, 2023-12-31:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2023-12-31 by unocelli