frangoteam/FUXA #922: [BUG] Circular gage scale

Joky12, 2023-10-03:
When i want to bind a tag with a circular gauge, and i set the min and max value different to 0, then it back to 0 when i click in the OK button.

[image]

Then i press OK and it's like:

[image]

When i open the circular gauge editor again, i see that:

[image]

unocelli, 2023-10-03:
Hi, thanks for the report

unocelli, 2023-10-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-10-11 by unocelli

Joky12, 2023-10-12:
Hi. The bug is resolved. I updated fuxa and problem is solved. Thank you!!