frangoteam/FUXA #923: Can I specify a specific page on the Events open page so that I can update the specified page when using the mixed container

fanhai, 2023-10-06:
Can I specify a specific page on the Events open page so that I can update the specified page when using the mixed container
[image]

fanhai, 2023-10-09:
实现思路：

First, divide the mixed container into several sub pages, each of which serves as a base page. You can place the base control page or drag and drop other pages to combine.

If you drag and drop the switch button directly on this page, the Events of the button can specify the basic sub page.

1）先将混合容器分成几个子页面，每个子页面作为一个基础页面，可以放基础控件页可以拖放其他页面进行组合。
如果这个页面直接拖放切换页面的按钮，按钮的Events可以指定基础子页面
2）如果这个页面直接拖放切换页面的按钮，按钮的Events可以指定基础子页面

unocelli, 2023-12-31:
We have added the Panel control, see if that solves your request

closed 2024-05-22 by unocelli