frangoteam/FUXA #933: createTag() / deleteTag() func

mariorl, 2023-10-18:
it will give a vast improvment to fuxa scripts I think

unocelli, 2023-10-19:
Hi, can you explain the use case please?

mariorl, 2023-10-20:
There are several scenarios, These are (for me) the two main ones:

- Create variables in runtime enables the end user (machine operator) to create trend graphs af any variable, in the way there's a generic trend created during design time by the system integrator, and at run time, the end user decide what to watch in this trend. This is very useful when need to find the best value for a parameter, so you can create a trend with that parameter and its consecuences to the machine behavior.

- Imagine you have a modbus-TCP ligth stack tower and you want to use each channel of that tower as a free indicator of a certain event or var state. In many factories, usually you will find a lot of these ligths piled up and mixed (Babel towers I like to call them). With this feature, it is avoided, the operator can decide what vars to watch.

- It will improve the faceplate feature (listed in Fuxa help as "how to use the same view").

- In many cases when the amount of tags in a project is huge, or when I need millisecond timestamp of an event, the poll mechanism becomes slow. One way to avoid it is to watch what variables changed or what events rised at PLC side, creating a small ring buffer in the PLC, and then only the new items of that buffer are exchanged with the supervision system. I have made this mechanism using python in the past. You can see this case as a sort of communication driver for special cases.

There are many other cases, think that this is very usual to use this feature with commercial HMIs (Siemens, Weintek, Kinco, etc). But I know that this, (to create these funcs) is not a simple task, as there are several issues concerning, such as safety, set limits to avoid creation of tons of not desired vars if there's an error during creation, etc. 

Also, it implies the machine operator need to know the PLC address of each var he want to watch, what is not the usual. the solution is actually that is the maintenence personnel who uses to create the list of vars, and the production personnel who watch that list, in a trend or somenthing like. Also, in certain scadas like Zenon, these addresses are visible during runtime if you have enough access rights.

Thanks, and again, congrats for that beautiful piece of software.