frangoteam/FUXA #935: Multiline Text with the Text tool

rosasor, 2023-10-18:
could the Text tool be modified to allow users to input multiple lines of text within a single Text box

andy8601, 2024-10-30:
> could the Text tool be modified to allow users to input multiple lines of text within a single Text box

Should add a part of the text editing tools, such as text alignment, line wrapping and so on...

@unocelli

mauroalexandre, 2024-11-06:
FUXA uses the SVGEditor JavaScript library. Sadly, multi-line text is not yet supported. 

See [the documentation](https://github.com/SVG-Edit/svgedit/issues/821) for more details.

unocelli, 2024-11-10:
Hi, a alternative solution could be to create a new html textarea control using <textarea> in a <foreignObject>

unocelli, 2025-07-09:
@tarleomkar Hi, as I have already said in #1850 I would use another approach by creating a new control