frangoteam/FUXA #937: Touch Screen Login, Keyboard required or simple PIN with Num Pad

unocelli, 2023-10-19:
### Discussed in https://github.com/frangoteam/FUXA/discussions/931

<div type='discussions-op-text'>

<sup>Originally posted by **RobertBudgen** October 18, 2023</sup>
Is it possible to enable the onscreen keyboard for Login.
Our system will be running on a touch screen in  Kiosk Mode, no Physical keyboard  or Windows keyboard  is available.
</div>

TomasAdvance, 2024-01-03:
Hi Fuxa Team,
it will be very helpfull if onscreen keyboard will be availabe for user Login.

unocelli, 2024-01-07:
This works in the latest version. let me know if you have any issues.

closed 2024-01-07 by unocelli

TomasAdvance, 2024-01-11:
Dear Fuxa team,
thanks for keyboard future in user login. I tested it and have two issue:
1) login dialog is close if you hit enter on the touchscreen keyboard. For example if you fill in user name throu touchscreen keayboard and hit enter on keyboard, Login dialog is close. Only posssible way how to close keyboard without closing login dialog is double click somewere on login dialog and then touchscreen keayboad is close without closing login dialog.

2) touchscreen keyboard is very small. Is it possible to change the dimension of keyboard?

commit 500b30ca (2024-01-18): add login dialog confirm to close #937