frangoteam/FUXA #943: Number Formatting for Very Small Floating Point Numbers

robbudge, 2023-10-26:
Currently running Version 1.1.16-1325
Some floating point numbers with scientific notation in the region of e-6 and less fail to render on both the output box and table object.

Raw Data Via OPCua
[image]
Dialog View
[image]

Even with extended formatting 10 Digits for the Output box and 0.000000 for the table we either get Zero or NAN.

unocelli, 2023-11-02:
Hi, sorry for the delay. What type of data is that? (Double, Float, ...)

robbudge, 2023-11-02:
That’s Float data.
The other image shows the raw data in the connections.
The data is OPCua of origin.

Best regards,
Robert Budgen
System Engineer Technical Solutions Expert

Bürkert Fluid Control Systems
Burkert Contromatic Inc.
5002 South Service Road<x-apple-data-detectors://4/1>
Burlington ON  L7L 5Y7<x-apple-data-detectors://4/1>
Cell:  (905) 630-0694<tel:(905)%20630-0694>
Phone: (905) 632-3033<tel:(905)%20632-3033>
Fax:     (905) 632-3833<tel:(905)%20632-3833>
***@***.******@***.***>
www.burkert.ca<http://www.burkert.ca/>

Sent from IPhone

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli