frangoteam/FUXA #946: How to install on embedded devices, armv8/armv7

liulujian00, 2023-11-02:
**Describe the bug**
1.Nodejs is installed on the arm and can be used normally.
https://nodejs.org/dist/v18.18.2/node-v18.18.2-linux-arm64.tar.gz

2.The downloaded tool kit can also be used normally.
https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu-a/10.3-2021.07/binrel/gcc-arm-10.3-2021.07-aarch64-aarch64-none-elf.tar.xz

3.A series of errors occurred when installing fuxa

npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797) npm ERR! code 1 npm ERR! path /usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-snap7 npm ERR! command failed npm ERR! command sh -c prebuild-install || node-gyp rebuild npm ERR! make: Entering directory '/usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-snap7/build' npm ERR!   CXX(target) Release/obj.target/snap7/deps/snap7/src/sys/snap_msgsock.o npm ERR! make: Leaving directory '/usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-snap7/build' npm ERR! prebuild-install WARN install No prebuilt binaries found (target=18.18.2 runtime=node arch=arm64 libc= platform=linux) npm ERR! gyp info it worked if it ends with ok npm ERR! gyp info using node-gyp@8.4.1 npm ERR! gyp info using node@18.18.2 | linux | arm64 npm ERR! gyp info find Python using Python version 3.8.6 found at "/usr/bin/python3" npm ERR! gyp info spawn /usr/bin/python3 npm ERR! gyp info spawn args [ npm ERR! gyp info spawn args   '/usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-gyp/gyp/gyp_main.py', npm ERR! gyp info spawn args   'binding.gyp', npm ERR! gyp info spawn args   '-f', npm ERR! gyp info spawn args   'make', npm ERR! gyp info spawn args   '-I', npm ERR! gyp info spawn args   '/usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-snap7/build/config.gypi', npm ERR! gyp info spawn args   '-I', npm ERR! gyp info spawn args   '/usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-gyp/addon.gypi', npm ERR! gyp info spawn args   '-I', npm ERR! gyp info spawn args   '/.cache/node-gyp/18.18.2/include/node/common.gypi', npm ERR! gyp info spawn args   '-Dlibrary=shared_library', npm ERR! gyp info spawn args   '-Dvisibility=default', npm ERR! gyp info spawn args   '-Dnode_root_dir=/.cache/node-gyp/18.18.2', npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-gyp', npm ERR! gyp info spawn args   '-Dnode_lib_file=/.cache/node-gyp/18.18.2/<(target_arch)/node.lib', npm ERR! gyp info spawn args   '-Dmodule_root_dir=/usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-snap7', npm ERR! gyp info spawn args   '-Dnode_engine=v8', npm ERR! gyp info spawn args   '--depth=.', npm ERR! gyp info spawn args   '--no-parallel', npm ERR! gyp info spawn args   '--generator-output', npm ERR! gyp info spawn args   'build', npm ERR! gyp info spawn args   '-Goutput_dir=.' npm ERR! gyp info spawn args ] npm ERR! gyp info spawn make npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ] npm ERR! make: g++: Command not found npm ERR! make: *** [snap7.target.mk:128: Release/obj.target/snap7/deps/snap7/src/sys/snap_msgsock.o] Error 127 npm ERR! gyp ERR! build error npm ERR! gyp ERR! stack Error: make failed with exit code: 2 npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-gyp/lib/build.js:194:23) npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:517:28) npm ERR! gyp ERR! stack     at ChildProcess.handle.onexit (node:internal/child_process:292:12) npm ERR! gyp ERR! System Linux 4.19.206 npm ERR! gyp ERR! command "/usr/local/node-v18.18.2-linux-arm64/bin/node" "/usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/.bin/node-gyp" "rebuild" npm ERR! gyp ERR! cwd /usr/local/node-v18.18.2-linux-arm64/lib/node_modules/@frangoteam/fuxa/node_modules/node-snap7 npm ERR! gyp ERR! node -v v18.18.2 npm ERR! gyp ERR! node-gyp -v v8.4.1 npm ERR! gyp ERR! not ok  npm ERR! A complete log of this run can be found in: /.npm/logs/2023-11-01T10_57_59_073Z-debug-0.log [root@ok3568:/mnt]# 

4. How to install on embedded devices, armv8/armv7？？？

unocelli, 2023-11-02:
Hi, the problem is the node-snap7 (driver for Siemens PLC).
You have 2 options:
- remove the node-snap7 from package.json on server
- use nodejs version 14.21.3

liulujian00, 2023-11-03:
This problem has troubled me for many days. Thank you for your suggestion. I followed your first suggestion and successfully installed FUXA. Thank you very much.

closed 2023-11-04 by liulujian00