frangoteam/FUXA #949: Bacnet Device connection error

diegomardev, 2023-11-06:
**Describe the bug**
Hi,
I was testing a bacnet device but I couldn't read anything, neither devices nor variables.
From the YABE program I can now read this device correctly.
Can you tell me what I should be doing wrong?
[image]
[image]
[image]
[image]

**Environment**
 - FUXA version: 1.1.15-1324
 - Node.js version: 14.21.3

Thanks!

Staybik, 2023-12-21:
Hi
I also have same problem with connection.

i looked in console and i’m guessing something is wrong with timeout becouse no matter wich value i set in webUI, An timeout error on console is displayed every 5sec.

So i’m guessing a node-bacnet is starting making connection but it cannot get whole data

unocelli, 2023-12-21:
HI,
I just tested with 'ProtoSense Technologies  BACnet/IP Device Server Simulator' and for me work as expected
you have set as interface the IP address where FUXA server runs?

[image]

diegomardev, 2023-12-27:
Yes I have configured the IP as seen in the screenshots or is it somewhere else where I should configure it?
Is it possible that you are using another version of node?
Thanks

unocelli, 2023-12-30:
I tested with v14.21.3

Staybik, 2024-01-03:
Hi
I’m doing config just like you
But also I’ve tried to change adpu timeout to 60000, change broadcast IP to 192.168.19.255, change pool time to 1,10,30sec and I still got communication error.

I don’t know why when i’m changing timeout or pool on web ui on console I still got timeout after 5sec no matter what time i set on ui
[image]

Thanks for your help !

nelumeh, 2024-01-31:
Has anyone found a solution to the problem?

acloaf, 2024-02-12:
I'm also experiencing the same problem.

I have exactly the same settings as you do; interface is the FUXA server IP, port 47808, adpu 60000. I had a machine running OrcaView go down and am trying to get a quick system in place until it's replaced.

diegomardev, 2024-02-22:
> I'm also experiencing the same problem.
> 
> I have exactly the same settings as you do; interface is the FUXA server IP, port 47808, adpu 60000. I had a machine running OrcaView go down and am trying to get a quick system in place until it's replaced.
> 
> Edit: I have been trying to run it in a docker container. I set up the network so that the docker container runs on the same subnet as the BACnet system. Just to test, I installed using npm on a different machine and BACnet immediately connected. My issue seems to be related to connecting to BACnet from a docker container.

I have tried to test this on a new machine without Docker with directly npm, but the result is the same, and it doesn't connect via BACnet. 
Has anyone managed to solve this?

unocelli, 2024-02-23:
Hi, is there a freeware simulator for pc so that the problem can be reproduced?

nelumeh, 2024-02-26:
> Hi, is there a freeware simulator for pc so that the problem can be reproduced?

Try to use **BACnet Simulator** from CBMS (https://www.cbmsstudio.com/downloads.html). 

In my case, the issue arises in the Linux system where FUXA is running. However, if I use the Windows system with FUXA installed, BACNet devices are properly detected.

unocelli, 2024-03-02:
Hi, fuxa uses the [node-bacnet](https://github.com/BiancoRoyal/node-bacstack) library, do you know of a working project that uses the same library?

nelumeh, 2024-03-11:
Hi, 

One of the solutions to the problem in the Linux system may be ignoring the address '0.0.0.0'. The changes should be made in the file /runtime/devices/bacnet/index.js:324. After such an operation, BACnet starts correctly, and it is possible to connect to devices.

[image]

[image]

unocelli, 2024-04-13:
I have updated the bacnet branch, which now becomes the reference branch for try to fix the problems

veodko, 2025-04-25:
Allegedly this should be fixed in v1.1.19 https://github.com/frangoteam/FUXA/releases/tag/v1.1.19 except that it wasn't. The pull request was indeed merged, but on a separate branch. Neither the 1.1.19 release or actual source code contain the fix to allow BACNet to work with any interface (0.0.0.0): https://github.com/frangoteam/FUXA/blob/master/server/runtime/devices/bacnet/index.js#L356