frangoteam/FUXA #954: bitmask

yoabau, 2023-11-08:
Hello everyone,

Thank you @IonuDragomir1  for your comment. It really helped me understanding the bitmask feature. The feature actually works as query of bit as defined [her](https://en.wikipedia.org/wiki/Mask_(computing)#Querying_the_status_of_a_bit). After selecting the required bit to monitor, the shape can return a color based on status (0=False or True=1) of one of the bits. Reusing your example I managed to get this result:
[image]

@unocelli As you can see "shape_10" returns green color since bit 0 and 1 are both True.
But I was expecting the function to return decimal value based on bit 0 and 1 regardless the status of bit 2 to 31 such as:
bit 0 | bit 1 | bit 2-31| value(dec) |
|-----|-----|-----|-----|
| 0 | 0 | x | 0 |
| 1 | 0 | x | 1 |
| 0 | 1 | x | 2 |
| 1 | 1 | x | 3 |

But it doesn't, and instead it seems to only check whether one of the bit has value 1=True?

_Originally posted by @yoabau in https://github.com/frangoteam/FUXA/discussions/295#discussioncomment-7392022_

unocelli, 2023-11-12:
Hi,
I think there are different interpretations of bitmask management and in any case it is not possible to satisfy everyone.
The bitmask is intended to be associated with the value of the the bit (1/0) and not with the value masked with the bitmask.

yoabau, 2023-11-12:
Hi @unocelli ,

Thank you for your reply. Let me develop some arguments to convince you that it is not a question of interpretation:

1. If I understand your explanation correctly, the feature leads to more confusion because:
   - Why is the returned value an integer without bitmask and just a bool (0 or 1) when one or __more__ bitmask activated?
   - Why is the returned value not a bool when no bitmask is activated?
   - What logic operand is used in above example above "shape_10" to return a bool? bit0 AND bit1, bit0 OR bit1, bit0 XOR bit1..
2. When opening Windows calculator in "programmer" mode and enter the following: 
   - 16#0F AND 16#11 = 16#01 -> 2#0000 1111 AND 2#0001 0001 = 2#0000 0001
   - 16#F0 AND 16#11 = 16#10 -> 2#0000 1111 AND 2#0001 0001 = 2#0001 0000
[image]

While the result is clearly an integer (encoded HEX or BIN), FUXA adds a second mystic logic operand to get only a bool? In other words, whatever the logic operand between the tag and the bitmask, the result must be of the same datatype:

| myTag | Logic Operand | myBitMask | Result | Comment |
| ------------- | ------------- | ------------- | ------------- | ------------- |
| INT  | AND  | INT | INT | correct |
| LONG | OR  | LONG | LONG | correct |
| INT  | XOR  | INT | BOOL | incorrect |

3. To compare to commercial HMI/SCADA system, programmers can use bitmask to display different result based on the value masked with the bitmask. E.g using the same case tested on Siemens WinCC Unified:
[image]
As you see, you can associate a bit number 2<sup>31...</sup> 2<sup>0</sup>
Here: 2<sup>0</sup> = WHITE, 2<sup>1</sup> = GREEN, 2<sup>2</sup> = RED whatever the bit activated between 2<sup>3</sup> and 2<sup>31</sup>

If FUXA aims to be a industrial web HMIs, SCADA, it needs to comply, at least, with industry standards such as IEC61131. Bitmask is actually a topic covered by IEC (free sample [here](https://www.fernhillsoftware.com/help/iec-61131/common-elements/functions-bitwise-boolean.html))

IonuDragomir1, 2023-11-13:
Hi,

I have encountered the same problem when using multiple bits on bitmask. For the same object (motor, pump, etc.) cannot set a color for each selected bit from bitmask. Please see my post [#865](https://github.com/frangoteam/FUXA/issues/865).

IonuDragomir1, 2023-11-13:
Hi,

I have encountered the same problem when using multiple bits on bitmask. For the same object (motor, pump, etc.) cannot set a color for each selected bit from bitmask. Please see my post [#865](https://github.com/frangoteam/FUXA/issues/865).

___________________________________________________________________________________

@yoabau, regarding questions you put on point 1:
1.1 When no bit is selected on bitmask it's ok to get the numerical value of a tag. For example, it's useful to change the color of a pump when its current is between 2 and 10 A.
When one or more bit(s) is selected on bitmask, the user can use either the boolean or numerical value of selected tag. It's true that when a single bit is selected, should be available only the bool value. But it can be easily selected.
1.2 When no bit is selected, the user gets tag's value and depending on it, the objected can be customized.
1.3 That's a very good question. When a single bit is selected, bitmask works as expected. The problem appears when multiple bits are selected as I wrote on [#865](https://github.com/frangoteam/FUXA/issues/865).

@unocelli, bitmask for only one selected bit work perfectly. My opinion is that when selecting multiple bits on bitmask, 2 behaviors (B1 and B2) should be available.

**B1:**
[image]
If the checkbox next to dropdown list is checked, the user should be able to select the logical operation to be made between the selected bits. Also, the + button should be disabled because the logical operation returns only one result so there's no reason to add another result for the same tag.

**B2**
[image]
If the checkbox next to dropdown list is not checked, next to each mask value should be a dropdown list which contains all the bitmask's selected bits. Then for each bit can be assigned a bool value.

Selecting multiple bits from a WORD or DWORD tag is often used in SCADA systems. Assigning a value to entire WORD or DWORD tag, when selecting multiple bits, is not always a solution because it could constantly change.