frangoteam/FUXA #955: [BUG]Bar charts do not show historical data

vifun, 2023-11-09:
**Describe the bug**
The data has been configured and saved。Bar charts do not show historical data

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Bar chart Settings'
2. Select 'date range'
3. The bar chart do not show historical data

**Screenshots**
[image]

**Environment**
 - FUXA 1.1.14-1208: 
 - Node.js 14.15.1: 
 - npm 6.14.8:
 - Platform/OS:
 - Browser:

unocelli, 2023-11-12:
Hi, thanks for the report, I will fix it as soon as possible

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli