frangoteam/FUXA #957: What's the usage of "Broadcast to client (frontend) all tags values" setting?

IonuDragomir1, 2023-11-10:
Hi,

Sorry if this question has been already put, but what's the usage of "Broadcast to client (frontend) all tags values" setting from System tab?

unocelli, 2023-11-12:
Hi, 
if active the backend sends all the values of the configured tags to the frontend regardless of the active view, otherwise it only sends those of the active views. However, it is recommended to deactivate the functionality for performance reasons, particularly if you have many tags.

IonuDragomir1, 2023-11-13:
Thanks for your reply, @unocelli!
Though I didn't understand completely your explanation. 
"if active the backend sends all the values of the configured tags to the frontend regardless of the active view, otherwise it only sends those of the active views." So if broadcast enabled,  the backend sends all the values of the configured tags to the frontend regardless of the active view. If broadcast disabled, what happens?

For example I have 10 tags and 2 views (v1 and v2). v1 has 3 output fields with 3 of the 10 tags. v2 has 7 output fields with the remaining 7 tags.
**Case 1 - broadcast enabled:**
If active view is v1, this means that backend sends to view (frontend) the value of the 3 tags to be displayed on output fields. The other 7 tag values are not send. Am I right?

**Case 2 - broadcast disabled:**
No mater which view is active, all 10 tag values are send from backend to frontend?

unocelli, 2023-11-14:
exactly but in the opposite, disabled sends only the necessary tags and enabled sends all

closed 2023-12-30 by unocelli