frangoteam/FUXA #959: [BUG] Device connection Modbus TCP

Bundyra, 2023-11-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Devices
2. Click on Add device
3. Add device ModbusTCP device
4. See Status "-"

**Expected behavior**
Device status "OK"

**Screenshots**
Device settings on FUXA version 1.1.5:
[image]
Device status:
[image]
[image]
Device settings on FUXA version 1.1.2:
[image]
Device status:
[image]

**Environment**
 - FUXA version: 1.1.5
 - Node.js version: 14.21.3
 - npm version: 6.14.18
 - Platform/OS: Linux Ubuntu 22.04.3
 - Browser: Opera

unocelli, 2023-11-12:
Hi, thanks for the report.
are there any error messages in the log file?

Bundyra, 2023-11-13:
The only errors i got in log file are:
[image]
And i got the same errors in console.

unocelli, 2023-11-14:
Nothing has changed between version 1.1.12 and version 1.1.15 for TCP. you have only more options for this connections. have you try version 1.1.14?

dougkite, 2023-11-15:
I had the same problem on 1.1.15-1324.  Modbus tcp connections seemed to work but no status or value data updates. I confirmed there was network traffic 
Reverted to 1.1.14-1243 and it works.

Bundyra, 2023-11-17:
On version 1.1.14 im getting error: "Cannot get /".

Bundyra, 2023-11-17:
After reinstalling everything and starting with new clean project i tried adding device. Still status "-" but in error logs i got something like this: 
[image]

Bundyra, 2023-11-17:
Tried now version 1.1.13 and got same issue (Status "-") and no values on tags.

Bundyra, 2023-11-17:
Just realized i have version 1.1.14-1243 installed on my server, and it works fine.

Bundyra, 2023-11-17:
Just realized that i have version 1.1.14-1243 installed on my server and it works fine.
Device configuration on this version:
[image]
Same device but on version 1.1.15-1324:
[image]
I changed polling rate to 10sec. so i dont get readMemory errors in the logs, no new errors status still "-" and no values in the tags.

unocelli, 2023-11-17:
Hi, the change between the two versions should not be relevant for the modbus. is the test on the same platform? are you able to debug it?

Bundyra, 2023-11-17:
Test is on the same platform but i am not able to debug it

unocelli, 2023-11-17:
Please tell me again what is your OS platform

Bundyra, 2023-11-17:
Linux Ubuntu 22.04.3 only in terminal version

unocelli, 2023-11-17:
you have installed from npm?

Bundyra, 2023-11-17:
I've installed from source files

unocelli, 2023-11-17:
I just tested it on Ubuntu 22 with nodejs v14.21.3 and a modbus simulator (ModRSsim2.exe), everything works as expected, try deleting and reinstalling node_modules

Bundyra, 2023-11-17:
I removed node_modules dir and package-lock.json file, but it didnt help

unocelli, 2023-11-17:
try connecting with another device like the simulator ModRSsim2...I have no idea, sorry

Bundyra, 2023-11-17:
I just installed fresh version but from npm. It works.

Bundyra, 2023-11-17:
Im gonna close as it works. Thank you for your time.

closed 2023-11-17 by Bundyra