frangoteam/FUXA #964: MQTT not publishing  on-change (or otherwise) of linked internal tags

lewsut, 2023-11-14:
We have the ability to set up a MQTT JSON Keyed payload, Brilliant!

[image]

We can create internal tags!

[image]

We can assign the MQTT keys to internal tags ...

[image]

This is where that ends for me, interestingly if i link to the MQTT status I get published topics.

[image]

however changing the values linked internal tags elisitis nothing that I can detect. I'm fairly sure this is my user error nut I have now tried everything, this seems like it should be fairly simple...

unocelli, 2023-11-16:
Hi, internal tags are only to be used on the frontend, they do not communicate with the backend and therefore cannot be published

lewsut, 2023-11-16:
It would perhaps be woth having them made unavailable from the backend selections then?

closed 2023-11-16 by lewsut

unocelli, 2023-11-16:
you are right, it is not consistent in all settings