frangoteam/FUXA #965: FUXA does not store tagname in InfluxDB.

zplorer52, 2023-11-15:
FUXA does store  "devicename=Fuxa_Connection_name" in InfluxDB1.8 which is difficult identify specific tag name. How to add tag name too?
[image]

Thanks.

unocelli, 2023-11-16:
Hi, thanks for the report.
I would be very grateful if you could test a small modification and tell me if works so that I can add it to the branch.
You have to modify in `fuxa\server\runtime\storage\influxdb\index.js` the line 106-109 with
` const tags = {
  id: tag.id,
  devicename: deviceName,
  name: tag.name || tag.tagref.name
}`

robbudge, 2023-11-23:
I get the same on InfluxDb2.0, would be nice to have  Device.Tag:Value
We use Grafana Alot and the UUID makes life hard. Nothing major though
I do think this would possibly cause issues with graphing as it looks that most items use the tags UUID as reference

unocelli, 2023-11-23:
Hi, the tag name is available in version 1.1.15 for influxdb 2.0

zplorer52, 2023-11-25:
> Hi, thanks for the report. I would be very grateful if you could test a small modification and tell me if works so that I can add it to the branch. You have to modify in [...]

> 1. I have added the line as proposed but couldn't see any differences immediately.
[image]

   2. After adding this line I've restarted the Influxdb1.8 service. And changed the "tagOptions" schedule and saved project in Fuxa connections. But it remains same.
   
[image]

[image]

unocelli, 2023-11-28:
Hi, then you can debug it...can you tell me on set a breakpoint on  line 94 what have the array tagsValues...could be that the attribute name are missing.

unocelli, 2024-05-22:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-05-22 by unocelli