frangoteam/FUXA #967: Script string issue.

lewsut, 2023-11-15:
In order to show a string as opposed to a number you have to start with a letter or some non number bases character. I wanted to show '22'+':'+'55' I will get the correct result in the console test but I will just get 22 once passed to the tag.

If a letter is placed in front of this, it's interpreted as a string e.g.   'x22'+':'+'55'  (x22:55 at tag)

Here are some simple examples.

[image]

[image]

lewsut, 2023-11-20:
@unocelli Did you change anything? Grabbed the latest docker (5 days old) and the scripts work as expected, I still have the old container and it doesn't work, as above

unocelli, 2023-11-23:
then it can be closed

lewsut, 2023-11-24:
Just checking as you never replied to "needs investigating" nor were there ever any updates on it...

closed 2023-11-24 by lewsut