frangoteam/FUXA #975: [BUG] working (connection || polling) overload!

zxywind, 2023-11-20:
**Describe the bug**
After starting yarn start, all OPC UA type device connections fail with the error "working (connection || polling) overload!". On the device page, all devices except the first node FUXA Server are listed as disconnected. Clicking "Edit" and then "OK" will reconnect the device, and everything will work normally after that. And, It seemed only OPC UA devices are  affected by this issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version: 1.1 
 - Node.js version: 16.2
 - npm version: 8.19.4
 - Platform/OS: rocky 9.2
 - Browser: edge

robbudge, 2023-11-23:
I also see this, Plus some other strange  errors

2023-11-23T18:02:47.853Z [ERR] 	'ns=4;s=|var|Edge Controller PIO3.Application.PVL.Config.LTun.Recipe.Transfer[0].Max_Volume' _startMonitor Error: Invalid Channel after performing transaction on CreateMonitoredItemsRequest
2023-11-23T18:02:47.909Z [ERR] 	'ns=4;s=|var|Edge Controller PIO3.Application.PVL.Config.LTun.Recipe.Transfer[0].Min_Flow' _startMonitor Error: Invalid Channel after performing transaction on CreateMonitoredItemsRequest
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 error listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit
(node:20) MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 finish listeners added to [WriteStream]. Use emitter.setMaxListeners() to increase limit

unocelli, 2023-11-23:
Hi, how many opcua devices have you configured?

robbudge, 2023-11-23:
Single Device, 900 tags

unocelli, 2023-11-23:
Have you try to separate the connection in more devices? there seems to be a limit of monitoring subscription, can be on FUXA but in opcua server too

zxywind, 2024-03-26:
I have separated the connection in more devices(split 4000 tags to 4 devices)
but It looks still working wrong.
 and reconnect is not working now.
thanks

unocelli, 2024-03-27:
Hi, it would be interesting to know on which opcua server and plc are you connecting, and also to know from how many subscriptions do you start having problems?

unocelli, 2024-05-22:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-05-22 by unocelli