frangoteam/FUXA #976: [BUG] Broken bool tags

lewsut, 2023-11-20:
**Describe the bug**
Can no longer toggle bool tags by any means from the front end. What was working before no longer works 

**To Reproduce**
Add a bool tag in Fuxa Server
Link a toggle button 
Toggle
Nothing changes

**Expected behavior**
A bool to change from true to false or vice versa 

**Environment**
    Docker
 FUXA version: 1.1.15-1356

unocelli, 2023-11-21:
Hi, Thanks for the report

lewsut, 2023-11-22:
I have checked I'm running the latest "fixed" code, this issue is not fixed at my end... same issue.

[image]

lewsut, 2023-11-22:
I guess the problem is 1 / 0 is getting changed to true / false and nothing is expecting that as I had worked on 1/0 at the front end

Not a lot of things can be set to "true" boolean denotation so a pipe set up to stop on 0 continues to turn or flow because I assume it sees boolean false as != 0. I'm not sure how you can implement both because at the moment they don't work well with each other , setting a toggle switch to be false / 1 or true / 0 causes issues. 

All I can think of is this and my coding is poor is
```

   if (type === 'number') {
       return parseFloat(value);
   } else if (type === 'boolean') {
       if (typeof value === 'string') {
           console.log(value.toLowerCase() !== 'false');
           return value.toLowerCase() !== 'false';
       }
       if (typeof value === 'number') {}
   } else if (type === 'string') {
       return value;
   } else {
       let val = parseFloat(value);
       if (Number.isNaN(val)) {
           // maybe boolean
           val = Number(value);
           // maybe string
           if (Number.isNaN(val)) {
               val = value;
           }
       } else {
           val = parseFloat(val.toFixed(5));
       }
       return val;
   }
   }
```

lewsut, 2023-11-22:
Using 
https://github.com/frangoteam/FUXA/blob/be9ee1d09b904345b7a55029bbfc3f84a0840f4e/server/runtime/devices/fuxaserver/index.js

Is the last time it works well, granted you can put a number into a bool... but at least it works as expected

unocelli, 2023-11-23:
Hi, first of all, considering that you are using fuxa server tags you can simply use number instead of boolean and it is also clearer since conditions are practically always with number. 
I can't confirm your issue. can you please send me a sample project?

lewsut, 2023-11-24:
I have emailed you an example. 

If I were to just use numbers what would be the point of having Boolean available? As you know it has its uses.

closed 2023-11-29 by lewsut