frangoteam/FUXA #978: [BUG] Changing view from layout button.

Bundyra, 2023-11-21:
**Describe the bug**
Layout button that allows 'view on click' function is not working properly when we redirect view to another 'subview' via button.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to view 'Test1' by layout 'view on click' function.
2. Click on button 'SUBVIEW' with open page function to view 'SUBVIEW'
3. Click on layout button 'Test1' with 'view on click function'.
4. See no action, we are still on 'SUBVIEW' page.

**Expected behavior**
Redirection to view connected to layout button with function 'view on click'.

**Screenshots**
1. Layout button with view 'Test1'
[image]
2. Page 'Test1' with button that opens page 'SUBVIEW'
[image]
3. Layout button doesn't open view 'Test1'
[image]

**Environment**
 - FUXA version: 1.1.15-1324
 - Node.js version: v16.21.3
 - npm version: 8.19.4
 - Platform/OS: Windows 10
 - Browser: Opera

unocelli, 2023-11-23:
Hi, Thanks for the report

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli