frangoteam/FUXA #980: [BUG] Caton

Beyond2324, 2023-11-23:
**Describe the bug**
Click on the properties to select the page to jump to or open the form program to directly pause.

Has the author encountered this situation before? Is it related to having too many tags? I have over a thousand tags and many pages that need to be redirected, which I have never encountered before

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version: 
 - Node.js version: 
 - npm version:
 - Platform/OS:
 - Browser:

unocelli, 2023-11-23:
Hi, 
certainly the performance depends on the platform where FUXA is installed, devices and protocols used. With more than a thousand tags I would disable the broadcast.
[image]

unocelli, 2024-05-22:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-05-22 by unocelli