frangoteam/FUXA #984: [BUG] Bar chart doesn't change date range.

Bundyra, 2023-11-24:
**Describe the bug**
Bar chart doesn't change date range (it's always shown for option 'Last day' group: 'hours'). It also shows wrong values.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Bar chart settings'
2. Select data range
3. Change data range for last week.
4. See no changes in the chart. See wrong values in the chart (values don't match with value from tag, in my example value in the chart
is greater than in the tag).

**Expected behavior**
Bar chart changes data range. Chart value matches value in the tag.

**Screenshots**
Chart settings (in English: Last week, group: Days):
[image]
Chart:
[image]
Tag value:
[image]
I'm sure that value in the tag doesn't drop, it can only rise so value in the tag should be highest possible. As we can see value on the chart is bigger than in the tag.

**Environment**
 - FUXA version: 1.1.15 - 1324
 - Node.js version: 16.21.3
 - npm version: 8.19.4
 - Platform/OS: Windows 10
 - Browser: Opera

Bundyra, 2023-11-28:
okay so maybe i am wrong in terms of values, but does the data range change works for you?

unocelli, 2023-12-08:
Yes

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli