frangoteam/FUXA #986: [BUG] Historical chart doesn't work with negative values.

Bundyra, 2023-11-24:
**Describe the bug**
Historical chart doesn't work with negative values. Chart is blank.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Chart settings'
2. Set historical chart option
3. Use chart with negative values
4. See blank chart

**Expected behavior**
Historical chart works and shows negative values.

**Screenshots**
Chart settings:
(English: Historical, last 8 hours):
[image]
Tag that i want to display:
[image]
Historical chart:
[image]
Normal chart:
[image]
Historical chart with **positive** value:
[image]

**Environment**
 - FUXA version: v1.1.15 - 1324
 - Node.js version: 16.21.3
 - npm version: 8.19.4
 - Platform/OS: Windows 10
 - Browser: Opera

unocelli, 2023-11-26:
Hi, I can't confirm the issue.
Are data from what kind of device?

Bundyra, 2023-11-28:
Its a modbusTCP connection with Solar-Log 2000

Bundyra, 2023-11-28:
Got it now, changed it from Input register to holding register and it works fine.

closed 2023-11-28 by Bundyra