frangoteam/FUXA #988: [FEATURE] Request for Live Update of Tags on Blur/Change Event

eduardojvieira, 2023-11-27:
**Feature Description**
In the current implementation of the FUXA SCADA project, the input component is designed to update the associated tag only upon pressing the 'Enter' key. This enhancement request aims to broaden this functionality by enabling the tag to update when the input field experiences a 'blur' (loss of focus) or a 'change' in value. This improvement is expected to yield a more dynamic and user-friendly experience, as it would allow the tag to be updated instantly whenever the user alters the input and shifts focus away from the field. This change addresses situations where users may forget to press 'Enter' after inputting their data, leading to unregistered changes.

**Proposed Solution**
The proposed solution involves refining the event handling of the input component to activate tag updates on 'blur' or 'change' events. An alternative approach could be the introduction of a 'live update' feature for the input tag, providing users the option to enable this functionality. This modification would require updates to the event listeners in the `html-input.component.ts` file, ensuring they respond appropriately to these new triggers and effectively update the tag with the new input values.

**Additional Context**
Implementing this feature would significantly enhance the interface's interactivity and efficiency, particularly in scenarios where immediate feedback or updates are crucial following user input. This change would eliminate the need for pressing 'Enter' to register data changes, aligning the application more closely with intuitive user interactions.

unocelli, 2023-11-30:
I quite agree with @lewsut. You can enable the dialog for input to be sure of confirmation
[image]
[image]

lewsut, 2023-12-08:
> Thanks for your response. I'll test the dialog mode. I think it will do the job. This was a request from the users, but I agree with you.

Listen to the users but never circumvent anything that makes them think about their actions. Users are lazy :)

unocelli, 2023-12-31:
I’m going to close this as resolved. let me know if you have any issues.

closed 2023-12-31 by unocelli