frangoteam/FUXA #994: [BUG] Not able to add data in data table at the run time

brijeshvpatel, 2023-12-03:
**Describe the bug**
I have created 5 tags from Modbus and it is working in Chart to show real time data but in data table I want to see each 5 columns and datetime to show realtime data at every interval set or tag changes. 

Even I can not add input value to data table at run time by clicking any button. I am not sure if that can be done bya a script as script manual is not descriptive.

**Environment**
 - FUXA version: fuxa 1.1.15
 - Node.js version:  16.14.15
 - npm version: 8.3.1
 - Platform/OS: Linux
 - Browser: Google Chrome

unocelli, 2023-12-08:
Hi,
I cannot confirm your problem, try sharing the project or an example

brijeshvpatel, 2023-12-12:
Refer attached project file and snapshots.  There are two issues;
1. Refer following pickup input screen. Whn you want to input any value in to data table, you have to click on refresh button instead of it just apear on the data table as soon as you change tag value from input value. This issue is with history data table. With realtime table it is not working at all to add data based on input value.

2. Refer history table screen. In sample project it is recording temperature where it is normal that user wants to see history table for specific period of time (let's say 1 week/day) but at the same time in history table it should also show real time values getting added. Which means realtime and history data must be in the same table view. 

[image]
[image]
[TempRecordHMI (7).json](https://github.com/frangoteam/FUXA/files/13646100/TempRecordHMI.7.json)

unocelli, 2023-12-12:
Hi, thanks for the descriptions.
I understand the request, however these are features and not bugs, please register them separately so that they can be managed.

brijeshvpatel, 2023-12-14:
Hi Unocelli,

It is a bug because if you can not add data in to the data table at the runtime without manual refresh, what is the purpose of just showing history data? Datatable has already a feature to show realtime data and because of so it should not show data only after clicking refresh button.  Please check realtime mode of datatable by showing data at the realtime from Tag change or from input data without clicking refresh button.

unocelli, 2024-11-18:
Is available in the latest version

closed 2024-11-18 by unocelli