frangoteam/FUXA #995: [BUG]

robbudge, 2023-12-03:
Testing  on Version 1.1.17-1370

When configuring Daq Connection Influx 2+,
Failure to Specify protocol in Server URL will break the runtime after a  'Restart' Command. Leaving no accesses to the system.
Daq Server URL 192.168.10.13:8086 would result in failure.
/server/_appdata/mysettings.json has to be manually edited too include http:// or https:// 
http://192.168.10.13:8086

Perhaps a validation check could be included to ensure https:// or http:// is included in the Daq Servers URL.

robbudge, 2023-12-08:
I was modifying the catalog setting on the V1.17 release and omitted the http:// prefix for my InfluxDB server. 

Obviously I got lots of errors. 
thinking it was a change error I restarted the Fuxa Process and it failed to launch and exited on startup. 
knowing what I had just changed, I modified the base file with the correct http:// prefix. 
restarted and all was good.

mauroalexandre, 2024-01-23:
#1062

unocelli, 2024-03-17:
This issue should be fixed in the latest version. let me know if you have any issues.

closed 2024-03-17 by unocelli