frangoteam/FUXA #998: No state shown for MQTT

23Errors, 2023-12-06:
Hi, I have a Mosquitto broker running on my PC. Connecting FUXA to the broker works fine accordig to the Terminal. But in the connections list there is no state shown. If I subscribing to a topic mosquitto shows the subscription of FUXA, but I never get values into FUXA over this subscription.

lewsut, 2023-12-08:
I have the same setup, it works fine for me... perhaps post your connections and setup details?  like this:

[image]

[image]

23Errors, 2023-12-08:
I have the same setup like you posted. The only difference is my adress. I am connecting the FUXA server to the Mosquitto broker over `mqtt://localhost:1883`

lewsut, 2023-12-08:
Just for fun, try using the IP address. Have you used something like MQTT Explorer to make sure everything is working as it should, Mosquitto can be a bit finicky to get working as intended (I find).

lewsut, 2023-12-08:
Show me your Mosquitto config please, I assume it has the following?

[image]

23Errors, 2023-12-11:
[mosquitto.txt](https://github.com/frangoteam/FUXA/files/13632065/mosquitto.txt)
Thats my config. I didnt changed much, but yes I changed allow_anonymous to true. But the listener wasnt configured by me, because the default port is 1883.

lewsut, 2023-12-11:
please try with the port specified...

23Errors, 2023-12-11:
I tried with specified port, but nothing changed. 😕

23Errors, 2023-12-11:
What I noticed is, that there is also no state shown in the FUXAServer. Is this normal maybe this is also an issue and I have to look on other settings than MQTT.

23Errors, 2023-12-11:
Ok, I tried the installation directly over npm. Before I was using the downloaded sourcecode and started FUXA with `npm start`.
And what should I say with the direct installation over npm it works flawless.

23Errors, 2023-12-11:
Ok, I tried the installation directly over npm. Before I was using the downloaded sourcecode and started FUXA with `npm start`.
And what should I say with the direct installation over npm it works flawless.
It was a stupid mistake in the installation process.
But anyways thanks for the help.

closed 2023-12-11 by 23Errors